betaflight/betaflight-configurator #1: Betaflight Configurator (Make users life easier)

borisbstyle, 2016-06-08:
@skaman82 
You mentioned a while back you were willing to help me out with betaflight configurator? I added you in the configurator dev team with full write access.

skaman82, 2016-06-11:
@borisbstyle Thanks. Any news on the betaflight logo? Do you have it in vector format?

blckmn, 2016-06-11:
@daggad from RCGroups gave me this link:
https://www.dropbox.com/sh/88fpe7x353aojto/AACwbM73vmyGABqAkeZhHfuba?dl=0

for getting some stickers made up.

skaman82, 2016-06-11:
@blckmn Thanks. Sadly it is not a vector format so I created a new logo I will use it if @borisbstyle likes it.

[image]

skaman82, 2016-06-12:
@borisbstyle Logo files for further use

[BF_LogoFiles.zip](https://github.com/betaflight/betaflight-configurator/files/310977/BF_LogoFiles.zip)
[image]

blckmn, 2016-06-12:
@skaman82 for the bee on the white background what does it look like with some black on its body?

skaman82, 2016-06-12:
@blckmn something like this?
[image]

borisbstyle, 2016-06-12:
Yeah that last bee looks better!

skaman82, 2016-06-12:
Ok, will update the configurator tomorrow

blckmn, 2016-06-12:
That's great. Awesome work.

skaman82, 2016-06-13:
Updated logo files:
[Betaflight Logo.zip](https://github.com/betaflight/betaflight-configurator/files/311451/Betaflight.Logo.zip)
[image]

JoeEurisko, 2016-06-15:
Awesome work,

Any way I can contribute to the development of the configurator?

skaman82, 2016-06-15:
@JoeEurisko of cours you can, how are your JS/C skills?

joehannouch, 2016-06-15:
@skaman82 Sorry just switched profiles,
I'm a game developer proficient in C++ and C# and long time web developer (front end).
I'm interested helping in both development and UX / UI as i've been a technical artist for quite some time.

Cheers

skaman82, 2016-06-15:
@JoeEurisko Great to hear. Could you maybe help to create a new backup/restore tab? See: https://github.com/betaflight/betaflight-configurator/issues/9

joehannouch, 2016-06-15:
@skaman82 Working on the backup/restore tab would be quite simple as an interface / css, but I would need some help when it comes to the functionality (backing up, saving prefs, etc.) where can i read about this? (i've swiftly read backup_restore.js, is that the way to go?)

nathantsoi, 2016-06-16:
dude, @skaman82. nice work. this looks way better than my ponies :)

RipperDrone, 2016-06-16:
Hi guys: Why don't you rename Betaflight to B-Flight or BFlight? Can be an acronym for BetaFlight, BorsFlight or Bee Flight which makes it fit to the wonderful logo that you created.

What do you think?

devinpearson, 2016-06-18:
@ripperdrone the name change was discussed at length in https://github.com/betaflight/betaflight/issues/422 and was decided that it should remain the same

Some ideas I would like to see is s way to write recipes/scripts  or common setups to reduce redoing some of the config each upgrade. More like the setting up sbus and turning on telemetry or modes like beeper and failsafe

Moving the commonly used tabs up the list and maybe a summary tab where anything you would tweak while tuning was on one page

RipperDrone, 2016-06-18:
@devinpearson understood, just thought I'd put my opinion in to try give merits to BorisB as the main promoter of BetaFlight by brnding his initial(s) into the product name.

Having a flex configurator which reads at runtime from a script style file which config elements and options to display is a very neat idea.

closed 2016-06-21 by borisbstyle

Yamaford, 2016-06-22:
Didn't really want to start a new issue for a suggestion but what do you think of adding a new tab for experimental settings that may or may not pan out? If they become permanent they could then be moved over to the proper tab. This would be a good place to look for new features without constantly modifying the other tabs once the dust settles.

borisbstyle, 2016-06-23:
there will be a pop up tab. No need to modify the existing one