betaflight/betaflight-configurator #9: Todo: add a backup function that works better than the current one

skaman82, 2016-06-14:
It should save everything through all profiles. At least all GUI stuff.
Making a backup/restore with a CLI is just bad and people do this just because there is no better option.

skaman82, 2016-06-14:
Here is my older concept layout how this could look like: https://github.com/skaman82/CF_GUI/blob/master/Layouts/CF0_Backup.jpg

Yamaford, 2016-06-14:
How about a menu with check boxes of the various sections in CLI to backup? Aux, Failsafe, RXrange, Colors, .....

skaman82, 2016-06-14:
@Yamaford not a bad idea, but that would make this process slow and I bet some users simply forget to check a checkbox and be angry afterwards :) I still think there should be two fixed backup models (STANDARD = only the GUI stuff including profile 0-2 and ADVANCED = complete CLI config).

skaman82, 2016-06-15:
@JoeEurisko backup_restore.js is a bit messy and a pain in the... but to work with. If you have a better backup architecture/stucture in mind, go for it. We just need have the old method as a fallback for older firmware versions. This are my ideas to make it better:

• Backup ALL profile settings
• Two backup methods (all gui settings / full cli config)
• If restoring, user can select areas to restore in a popup (RX, LED, RATES, PIDs....)
• Backups are stored in configurator BUT
• Also can be exported as a file, restore from a exported file should work as well
• Version info and optional compatibility warnings

joehannouch, 2016-06-15:
Ok great.  Please tag @JoeHannouch instead of  @JoeEurisko thanks

skaman82, 2016-06-15:
@JoeHannouch by the way, there is already a icon in place for the backup tab :)

KiteAnton, 2016-06-15:
Is there really a need to have the configurator compatible with older fw? Backward compatibility can sometimes restrict development and could it not sometime just be acceptable to say you need at least fw version X to use this?

skaman82, 2016-06-15:
I think so, at least at the moment. Still know many pilots flying BF 2.5.x. It is always bad to demand users to upgrade. I also don't see how the fallback would restrict the new functionality. Keeping the old approach just makes the code size a little bit bigger and it can be abandoned after a while.

skaman82, 2016-06-15:
I am not sure if betaflight has ported this already, but i think this could help: https://github.com/cleanflight/cleanflight/pull/1049

borisbstyle, 2016-06-21:
I think we need to add the newest available parameters to backup / restore function.

mikeller, 2016-06-21:
@borisbstyle: Either that, or wait until I add buttons to automatically save / restore a CLI dump without having to touch the CLI. ;-)

I am working on a way to create a CLI 'diff' showing only settings that are not default values. Once this is done, I'll look into making creating / restoring CLI dumps a bit more comfortable in the configurator.

borisbstyle, 2016-06-21:
@mikeller 
Ok curious to what you will have.

Are there any other commands we want to add in MSP? I mean 2.8.2 we can dedicate more to integration within configurator?

borisbstyle, 2016-07-13:
@mikeller 
Yeah that name sounds better indeed.
Automatically pickup up the values would be the best thing!

There will be some new parameters to come soon. From next versions the PID int will stay as it is and will be called legacy pid controller. Will also not be changed much. The Float PID will become betaflight pid controller and thats where all the changes will be done in the future.

borisbstyle, 2016-07-13:
btw backup / restore feature is really wanted!

mikeller, 2016-07-13:
> btw backup / restore feature is really wanted!

Good point, will give that one priority.

Austinthemighty, 2016-08-21:
I will try and help on this one to, may not be able to do much due to school, but i will try, also in my opinion we should have it be backwards compatible, but add all of the new parameters to it, while not overwriting the old ones.

mikeller, 2016-08-21:
I've already started working on this, it's going to be based on using a CLI dump / diff as a backup, which should make it a bit more portable across firmware updates.

Austinthemighty, 2016-08-22:
ok can you send me the repo link

closed 2018-05-13 by stale[bot]