betaflight/betaflight-configurator #21: Add Yaw Rc Rate to the PID Tuning Tab together with Roll/Pitch Rc Rate

borisbstyle, 2016-06-19:
Add yaw rc Rate from the new MSP commands (available in 2.8.1)

Probably also some check needed to verify if 2.8.1 betaflight version or higher is actually connected? Normally you would check for MSP version, but that one is the same yet to keep compatibility

mikeller, 2016-06-19:
A lot of places in the configurator check for the actual version of the FC firmware, which is probably a better way to check, as it allows to make a distinction even for changes that are not tied to a change in protocol.

And I think it would be nice to keep compatibility with cleanflight MSP as much as possible, as this will allow for the use of configuration utilities on platforms that cannot run chrome apps (i.e. android, iOS).

borisbstyle, 2016-06-19:
Yeah I agree

BTW just make PR's into the master from now. No need for dev branch as the changes will be small and a lot of updates in the beginning

skaman82, 2016-06-20:
@mikeller could we reduce the height of the two graphs at least to 80% to the current height (change the proportions)? Currently the TPA field is way to far down.

closed 2016-06-21 by borisbstyle