betaflight/betaflight-configurator #24: Add rate calculator // Or make the Rate curve realistic

borisbstyle, 2016-06-19:


mikeller, 2016-06-19:
The current rate curve is what was displayed in the receiver tab, so it probably is what it should be for < 2.8. My plan is to add a curve for your new rates, and make it display one or the other, depending on the version.

However, I think instead of me trying to untangle the formula that you've used to make the rates diagram work in the XLS spreadsheet, it would be way easier if you helped me with the code to draw the curve in the configurator. This is what is used to draw the curve currently:

https://jsfiddle.net/uq98vj3r/

borisbstyle, 2016-06-20:
@mikeller 
Thanks for the link. I will add that one!

mikeller, 2016-06-20:
Hehe, that's not quite what I had in mind. The above jsfiddle is an extract of the code in cleanflight-configurator that is used to draw the throttle curve. My request was for you to modify it in such a way that it adequately reflects betaflight's new rate curve, and I'll put it into the betaflight-configurator. This saves me having to reverse engineer the calculations for the rate curve from your spreadsheet or the FC code.

borisbstyle, 2016-06-20:
I know that but I was saying thanks for the link so I can easier visualize
what I am doing :)

skaman82, 2016-06-20:
@mikeller could we reduce the height of the two graphs at least to 80% to the current height (change the proportions)? Currently the TPA field is way to far down.

mikeller, 2016-06-20:
@skaman82: Done in #27. Could you have a look into how to best make the marked grey areas in the screenshot below look white? The intention is to make it more obvious that the single value for rc rate / rc expo aplies to multiple rows.

[image]

@borisbstyle, @skaman82: I was thinking about moving the throttle rate / expo settings onto the PID tab as well, and then moving the profile and rateprofile selectors to the top of the PID tab. This would create a very full PID tab, but IMO having to switch between multiple tabs in order to set up different profiles / rateprofiles is very cumbersome. Your thoughts?

borisbstyle, 2016-06-20:
@mikeller 
Yes I agree that the most touched values should be in one tab. Throttle stuff should be more below, but really not the most important now.

skaman82, 2016-06-20:
Ok, I take a look at this tomorrow.

skaman82, 2016-06-20:
The profile selector is already in the header of the configurator. So it is accessable from every tab. 

Honest question: what for so we need the rate profiles for? I never used it and find two different sorts of profiles very confusing.

borisbstyle, 2016-06-20:
I totally agree about those profiles. But some users use it. Once you
create something its hard to take away. I would have never seperate those

KiteAnton, 2016-06-20:
I agree also about the profiles/rateprofiles but last time the profile and rateprofile where merged there where some complains so it got reverted.

borisbstyle, 2016-06-20:
Yes I know :D. Some really got used to it. I never understood why should
you have rate profile. It is easier to change it on the field than having
some confusing switches

mikeller, 2016-06-20:
@skaman82: The profile selector only affects the settings on the PID tab, and the throttle settings that are for now on the receiver tab. Once the throttle settings have been moved, everything that is affected by profile selection will be on the PID tab. In this case it is a lot less confusing to have the profile selector only on this tab, and users are no longer left wondering which values on the other tabs are affected by profile selection.

Re rateprofiles, I think the only rationale for keeping them around is that they can be switched in-flight (i.e. when conditions change). If the profiles could be made switchable in-flight, without requiring an FC reboot, then the rationale for rateprofiles would be pretty much gone.

skaman82, 2016-06-20:
ok, this makes sense (i didn’t realized that modes/adjustments are not affected anymore by profiles in betaflight)

mikeller, 2016-06-20:
@borisbstyle:

I've added your formula to the curve calculation:

https://jsfiddle.net/uq98vj3r/6/

It seems to give me very extreme results, let me know if this is how you want it.

skaman82, 2016-06-22:
@mikeller this should help. See https://github.com/betaflight/betaflight-configurator/commit/47ec02932198e52e1d82c469dae66320a5271969

[image]

mikeller, 2016-06-22:
@skaman82: Looking good, thanks!

borisbstyle, 2016-06-24:
@mikeller 
Doesnt look really right. I will have a look soon

mikeller, 2016-06-24:
Thought it looked weird. But no hurry, I think sorting out 3.0 is more important, there seem to be a lot of people waiting on that.

borisbstyle, 2016-06-24:
RC rate is not being accounted at all. Try making rc rate 1.0 and rate 0. You will get a flat line.

I will have a look

mikeller, 2016-07-05:
@borisbstyle: I think this is about as good as we're gonna get for the curves in #79.

closed 2016-07-06 by borisbstyle