betaflight/betaflight-configurator #32: Stop configurator from breaking for users with firmware <= 2.8.1-RC1

mikeller, 2016-06-20:
A lot of the new features that have been added (#25, #26, #31,...), the functionality to support users has been removed, leaving the configurator broken and unusable for users that have not yet updated to the latest version.

We should either:
- re-add the removed features, and properly version - gate them, so the users get the UI that matches their FC version
- add code to prevent users from connecting with an old version of betaflight

Just leaving it the way it is will lead to a bad user experience.

borisbstyle, 2016-06-20:
@mikeller 
The configurator will simply not load tabs for some versions. No big deal if you ask me. pre 2.8.1 its CF configurator and after that versions will have to be maintained.

Whats the point of using betaflight configurator for old versions anyway?

skaman82, 2016-06-20:
1+ for keeping it compatible with older versions

mikeller, 2016-06-20:
@borisbstyle: The point is giving users a good user experience. At the moment, the vast majority of users of betaflight probably have not migrated to the latest version. If they try out the betaflight configurator (maybe because they want to update to latest), they will experience it as broken, go back to cleanflight configurator, and probably never try again.

If we at least present them a dialog that tells them: 'This tool only supports the latest version, your version is too old, please update your firmware before connecting', then they will not get the disappointment of using something that is broken, and the risk of breaking their configuration by doing so.

borisbstyle, 2016-06-20:
Well my point is that there is nothing extra they can configure when using the old version? It is practically the same as standard CF. 
They still can access the flash page to flash the latest version.

skaman82, 2016-06-20:
Yes you are right, but if some users lets say have multiple quads with different versions of BF they mostly woud like to change settings with BG configurator and not switching between configurators all the time.

mikeller, 2016-06-20:
@borisbstyle:

> Well my point is that there is nothing extra they can configure when using the old version? It is practically the same as standard CF. 

In  #25 and  #26, a number of settings that used to be available in CF have been removed, so they will not be able to control them any more outside of CLI.

Also, there is a bunch of tabs that do not work any more because of the MSP being unsupported, so they can not configure anything that's on them.

Just leaving it in a state that appears to users as being broken will turn them away from using the betaflight-configurator, or, even worse, will have them open issues for things we already know are broken.

Hence my suggestion to just not allow them to connect (and yes, flashing would still be supported).

skaman82, 2016-06-20:
If we go with not alowing to connect  with older versions we should also not allow to flash older versions because the would be not compatible with this configurator.

It would be a living hell for users if we would say: you can flash any older version with BG configurator but you have to switch to another configurator to change settings.

borisbstyle, 2016-06-20:
@skaman82 
Yeah probably good idea to prevent flashing the older versions!

mikeller, 2016-06-20:
@skaman82: 'Not allow' is a bit difficult, as they can always choose to download or build any version, and then load it as a file / flash it, or even flash something that is not betaflight. But we definitely should not offer them old and unsupported versions when we present them a selection of versions to choose from.

skaman82, 2016-06-20:
That is exactly what i ment :)

KiteAnton, 2016-06-21:
I think it will be a lot easier to develop and maintain if we only support newer versions of betaflight for the configurator. Earlier versions can always use the cleanflight configurator. But I agree that if we limit that then we should also limit the possible betaflight versions to be downloaded online in the firmware flasher page.

mikeller, 2016-06-21:
It will be easier for now, but that won't last for long as going forward, we will have to support all the versions between now and then again, and will be required to feature gate new stuff that gets added over time.

If we decide to do this, we should lock out users of unsupported versions. Providing an experience that is known to be broken is only going to lead to frustrated users and unnecessary support requests.

borisbstyle, 2016-06-21:
I havent been doing much with the configurator till now, but now I see at
the way how is set up it doesnt make sense to me that stuff hangs on wrong
versions.

It would totally be possible to give feedback to user with a pop up or
somehow when certain option is not available on his firmware.

I prefere also users to have the latest firmware version anyway as it makes
support task much easier

mikeller, 2016-06-21:
> I prefere also users to have the latest firmware version anyway as it makes
support task much easier

Makes total sense, after all, the firmware tries to be bleeding edge.

Will look into adding a popup.

borisbstyle, 2016-06-21:
@mikeller 
Ok I changed the name to add some backwards compatibility. I think its not a hard task to add back old parts for versions older than 2.8.1. We will just dont care about the RC's.

mikeller, 2016-06-21:
Then we should start to do proper RCs, and not just willy nilly throw new features in between RC and release. Or just give every release a number of its own and stop calling it RC when it's not...

borisbstyle, 2016-06-24:
I suggest to work like this for checking the versions
`if (CONFIG.flightControllerIdentifier == "BTFL" && semver.gte(CONFIG.flightControllerVersion, "2.8.1")) {`

borisbstyle, 2016-06-24:
Done. It is backwards compatible now with all versions. We need to add this in each PR now

closed 2016-06-24 by borisbstyle