betaflight/betaflight-configurator #35: Unable to flash 2.8.1

bbbenji, 2016-06-22:
Posting for posterity.

Using BFC 1.2 I was unable to flash my SPRacingF3 board from 2.7.1 to 2.8.1. Using the same `Manual baud rate` and `Full chip erase` options in CFC, it went without a problem. Issue occurred if I downloaded the fw via BFC. I did not attempt to flash the file downloaded directly from BF repo via BFC, only via CFC. Issue may lay here. I probably wont be testing this problem anymore.

borisbstyle, 2016-06-22:
I actually successfully flashed mine on that way

bbbenji, 2016-06-23:
I will try to do some testing this weekend and post more info. If anyone else can confirm this working i might as well close it.

KiteAnton, 2016-06-23:
I just tried flashing 2.8.1 downloaded online via the configurator to a SPRacingF3 board and it worked. The SPRacingF3 board has always been a bit problematic to flash, you need to set manual baud rate and try to unplug/plug in the board some times before it works. That is however not an issue with either betaflight configurator (it is the same in cleanflight configurator) nor betaflight FW.

closed 2016-06-24 by borisbstyle