betaflight/betaflight-configurator #36: Add ability to change min_check/max_check

compudaze, 2016-06-22:
Would be nice to be able to change min_check and max_check in the configurator. # # @

borisbstyle, 2016-06-22:
Why? Its not that important to me

compudaze, 2016-06-23:
It's the only thing left, sans filters (which are coming to the configurator I thought), I goto the command line for since the new betaflight configurator was made available.

TiTidom-RC, 2016-06-28:
+1 it would be great to have them (min & max check) in the GUI. 

Always fewer parameters for which I need to go into the CLI :)

CBLimerick, 2016-07-05:
+1. I found this recently and was surprised its not more requested. To me its important to set your remote to do full range of 1000 to 2000 and then min & max check to 1000 and 2000 .

closed 2018-05-13 by stale[bot]