betaflight/betaflight-configurator #38: Some Interface Elements not working with DogeFC

SSPW, 2016-06-23:
With DogeFC flashed 2.8.1;
ESC/Motor Protocol always shows PWM, but in CLI it is ONESHOT125
Gyro Update Freq and PID Loop Freq are blank

Changing either of the above and saving has no effect, confirmed with cli dump.

These ui elements work with a NAZE with 2.8.1, but not DOGE

borisbstyle, 2016-06-23:
It does work on my DOGE for sure.

Are you using official 2.8.1 or some of the pre releases?

SSPW, 2016-06-23:
# version
# BetaFlight/DOGE 2.8.1 Jun  8 2016 / 13:28:52 (c0b00fd)

KiteAnton, 2016-06-23:
@SSPW The official Betaflight 2.8.1 was released 4 days ago and your version is from June 8th, you probably are running RC1 or RC2. Just update to the latest official release and it will work.

closed 2016-06-23 by KiteAnton