betaflight/betaflight-configurator #40: [Idea] Firmware Flasher, change firmware selector

bilson, 2016-06-24:
In firmware flasher tab, firmware selector is very long and with version 3.0, each new version adds ~30 new targets. I think is a good idea change to two select box, one with firmware version, and second with target list.

borisbstyle, 2016-06-24:
good idea indeed

skaman82, 2016-06-24:
I think ist would be more helpful the other way around. First select target then select all the versions

blckmn, 2016-07-02:
I think @skaman82 is right. Target, then version.

It would be good to get v3.0.0 in there now as unstable for testing ? thoughts @borisbstyle @martinbudden

mikeller, 2016-07-03:
@blckmn: Sounds like a good idea. Maybe have a checkbox 'Show pre-release versions', and present the user a 'be aware that these versions may fail and cause damage, do bench test first' dialog?

closed 2016-07-05 by bilson