betaflight/betaflight-configurator #42: Add ability to enable / disable Senors

borisbstyle, 2016-06-24:
User should be able to turn on or off sensors in GUI

KiteAnton, 2016-06-24:
I had an idea that you could perhaps just press the sensors in the top bar to toggle them on/off? Or would that be to "hidden" since people are not used to that functionality?

skaman82, 2016-06-24:
I think this would be better placed in the config tab.

borisbstyle, 2016-06-24:
@KiteAnton 
I was actually thinking the same, but it wont be intiutive.

I did some MSP preparations. Can you perhaps add this into config tab @KiteAnton ?
https://github.com/betaflight/betaflight-configurator/commit/3b496324faca2de4921acb136132efd572d214eb

I want to continue with some more compatibility stuff to be added and try to finish it to bring out the updated configurator. It would be nice if the sensors also could be disabled / enabled.

KiteAnton, 2016-06-24:
Ok maybe it can be both. I will try to get some time during the weekend to add this.

closed 2016-06-25 by borisbstyle