betaflight/betaflight-configurator #43: Cleanflight-version specific code must be migrated

gaelj, 2016-06-25:
`msp.js` and other files contain conditional code related to api version.
This code is still adapoted to Cleanflight, whose MSP version is not synced with Betaflight's.
This will yield unexpected behaviour, especially when BF's API version is increased.

borisbstyle, 2016-06-25:
not at all actually. All betaflight specific commands are only applied to
betaflight firmware. All old betaflight firmware will also be supported
from next release.

MSP does a check for firmware version and type

gaelj, 2016-06-25:
@borisbstyle what about these blocks for example?
Line 450:
```
if (semver.lt(CONFIG.apiVersion, "1.18.0"))
    MISC.mag_declination = data.getInt16(offset, 1) / 10; // -1800-1800
else
    MISC.mag_declination = data.getInt16(offset, 1) / 100; // -18000-18000    
```

Line 468:
```
if (semver.lt(CONFIG.apiVersion, "1.17.0")) {
    offset += 2;
    _3D.deadband3d_throttle = data.getUint16(offset, 1);
}
```
If unchanged, they will alter the configurator's behaviour with future releases of BF MSP.
Don't you agree ?

borisbstyle, 2016-06-25:
Thats all fine. That cant hurt anything.

Betaflight supports all those apis.

There is:
MSP version
Firmware identifier
Firmware version.

Betaflight and Cleanflight share same MSP versions. That helps supporting
OSDs and mobile apps like ezgui.
Only Betaflight has some additional commands only supported by betaflight
configurator.

And everything is backwards compatible.

gaelj, 2016-06-25:
In thise case in my LED strip PR, which currently uses MSP version 17, needs to be updated to jump to MSP version 20, which is the CF MSP version where all current new LED functions are present.
But this means the bahaviour in lines 450 and 458 (those are just examples) will change.
What do you think ?

borisbstyle, 2016-06-25:
@gaelj 
You need to inspect differences between MSP 17 to 20. It may be there is much more than just leds

gaelj, 2016-06-25:
Got it.

gaelj, 2016-06-26:
#### MSP changes from 17 to 20
Current CF version is 19 ; I have a pending LEDstrip improvement PR which will bump it to 20.

##### CF MSP 1.17
- move `deadband3d_throttle` from `MSP_3D` to `MSP_RC_DEADBAND`

##### CF MSP 1.18
- `MSP_MISC`: `mag_declination` / 100 instead of / 10 (https://github.com/cleanflight/cleanflight-configurator/pull/369).
- `MSP_PID`: remove float conversion of PID values
- function: MAVlink telemetry https://github.com/cleanflight/cleanflight/pull/806
- function: LEDstrip: GPS, Blink, RSSI

##### CF MSP 1.19
- `MSP_LED_STRIP_MODECOLOR`
- function: LEDstrip: battery

##### CF MSP 1.20
- `MSP LED_STRIP_CONFIG` now uses new 32 bit format for each LED definition
- function: LEDstrip: Larson scanner, blink on landing, RSSI is now a function instead of an overlay, Blink is an overlay instead of a function

#### Actions
1. I issue 4 distinct BF PRs to merge CF changes in MSP verisons 17 and 18 (including MAVlink telemetry, I hope I'm not getting myself into trouble here -> if too complex, maybe we leave it aside ?)
2. bump MSP to 20 in this LED strip PR and the related configurator PR
3. finally, maybe we wait until the MSP 20 changes are merged into CF before merging them here, just to be safe, and polite (I don't want anyone to feel I am forcing changes to one fork by pushing them to another)

How does that sound to you @borisbstyle ?

gaelj, 2016-06-26:
>    move deadband3d_throttle from MSP_3D to MSP_RC_DEADBAND

The main change here was to put this setting in profile instead of main config. Do you want this ? Or would you just want the MSP frames to be compatible ?

skaman82, 2016-06-26:
@gaelj @borisbstyle regarding 3: I don't think it would be bad at all to merge this new LED function here first. It is summer and people are flying NOW. So every day counts to get it out here for users to benefit from it. Since the main goal of CF 1.14 is the OSD support, it can take a while till it will be ready and I am almost sure that Dominic will merge his OSD code first and all other PRs have to be updated/rebased till they could be included as well.

borisbstyle, 2016-06-26:
We dont want it to break something else.

This change changes MSP version and that impact of that needs to be checked.

skaman82, 2016-06-26:
Yes sure. My comment was just related to Gäels question about waiting for it o be merged in CF.

gaelj, 2016-06-26:
If I only modify the MSP frame format and not the underlying changes, We could have something compatible quite quickly.

borisbstyle, 2016-06-26:
but what if those users use cleanflight configurator?

skaman82, 2016-06-26:
From now on we schould assume that users would use the bf configurator for betaflight otherwise the development of new stuff would still depend from CF configurator. If i am not mistaken this is the whole point of the bf configurator to change that?

gaelj, 2016-06-26:
> but what if those users use cleanflight configurator?

What I meant is, I move deadband3d_throttle from MSP_3D to MSP_RC_DEADBAND. So CF configurator will indeed work. The difference being, (unless I missing anything) BF will still see at as a global parameter. The CF commit is this one https://github.com/cleanflight/cleanflight/pull/1976/commits/b6830dea314b7603590f28d11bc79c1ae43041ad.

> From now on we schould assume that users would use the bf configurator for betaflight otherwise the development of new stuff would still depend from CF configurator

Not entirely true regarding MSP protocol, as it is shared with other devices and frame format should indeed be kept coherent between forks. Handling different MSP versions is one thing, adding the granularity of which Flight control's MSP it was issued by would be a nightmare. What BF developers are allowed to do, is add their own additional MSP messages, and this is what the specific configurator is for.

skaman82, 2016-06-26:
> What I meant is, I move deadband3d_throttle from MSP_3D to MSP_RC_DEADBAND. So CF configurator will indeed work. The difference being, (unless I missing anything) BF will still see at as a global parameter. The CF commit is this one cleanflight/cleanflight@b6830de.

Ok, got it. Thanks.

gaelj, 2016-06-27:
Just discovered 

> MSP_MISC: mag_declination not scaled by 10 

 got left behind after a code cleanup.
MAVlink telemetry is just a serial port in the configurator, I think it can be skipped for now.
MSP PID is already in BF too.
I put back MSP 20 instead of 17 in the LED code so we should be ready.

closed 2016-07-21 by gaelj