betaflight/betaflight-configurator #49: Adjustments

micworg, 2016-06-26:
Hi, the adjustments in the Betaflight Configurator are still the ones from Cleanflight. Some of them are not available in Betaflight (yet?).    

(BTW: What do you think about this: http://www.rcgroups.com/forums/showthread.php?t=2681167 ? I have done this for Betaflight too. Maybe this can become part of Betaflight

KiteAnton, 2016-06-26:
Seems to be some people requesting telemetry feedback for actual adjustment values so if you can put the solution together in a PR towards betaflight then it can be reviewed/discussed there.

KiteAnton, 2016-06-27:
@micworg can you continue the discussion on the betaflight page instead, see this [#409](https://github.com/betaflight/betaflight/issues/409), it is a similar discussion.

closed 2016-06-27 by KiteAnton