betaflight/betaflight-configurator #51: Help people be aware of the release notes

cmavr8, 2016-06-26:
I think most of the people around here understand that it's important to read release notes, especially when flashing beta software (such as BF) and when major version bumps occur.

I also think that **introducing the BF configurator** is a great development but it **makes it easy to flash BF without reading the notes**.

While using the CF configurator, the user has to browse to the BF releases on github, scroll through the notes and grab the hex file. With BF configurator the user can merely install the configurator and flash from the drop down menu, without ever looking at the release notes.

**I suggest that the configurator features some measure to "force" new users to read the notes.** 

Maybe when the user loads a firmware, they should be shown big flashcards for each and every release note (one by one, with an OK button).

It goes without saying that this should be done once per release and there must be an option to completely disable this feature so that the pros who flash many boards are not hindered.

4 people have thumb-uped this idea, including Joshua Bardwell and Jesse Peacock, as a [youtube video comment](https://www.youtube.com/watch?v=XevARP37qOo).

mikeller, 2016-06-26:
+1 from me. Could probably even be auto-generated from GitHub release notes.

borisbstyle, 2016-06-26:
The release notes are shown when you load the online hex in the
configurator actually. Maybe some additional big font is needed in the
flash window as extra warning to make sure they read it.

Also flashing standard repos is always safe. There is a special checkbox
for unstable releases.

mikeller, 2016-06-26:
> Also flashing standard repos is always safe. There is a special checkbox
for unstable releases.

I think this is not only about stability, it's more about features and config settings that get renamed and defaults that get changed.

There is a lot of 

> Q: 'feature X doesn't work as expected any more after update' 

> A: 'this is because the config for feature X has changed, you need to modify your config, see release notes'

going on in here. If we can reduce this by getting people to read the release notes when updating it would free up devs to do more productive work instead of answering this kind of question.

skaman82, 2016-06-26:
I don't think that would realy help. Some people just don't want to read the notes. Even if you show the release notes in 30px bold.

borisbstyle, 2016-06-26:
they dont read anyway. I actually sometimes get same quastion every day and
sometimes multiple times per day :)

mikeller, 2016-06-26:
@borisbstyle:

> they dont read anyway. I actually sometimes get same quastion every day and
sometimes multiple times per day :)

You got a point there.

I hope that, once my CLI 'diff' code is in the firmware, and we've got diff / restore as buttons in the configurator, this will at least remove the problem of changing default values across updates...

cmavr8, 2016-06-27:
> I don't think that would realy help. Some people just don't want to read the notes. Even if you show the release notes in 30px bold.

Absolutely true, but wouldn't it help some people in the middle? The ones that may forget (or not yet know) about release notes, but would still read them if they popped up in front of them.

It could be that these people are too few to be worth saving. Your guys' call. You'll pay the price (dev time) so you decide. 

Cheers :)
And thanks for all the great work to all.

skaman82, 2016-06-27:
I think that the notes are clearly shown already

blckmn, 2016-07-02:
Should the firmware flasher front page have a release notes that changes when you select the firmware?

mikeller, 2016-07-03:
I would be in favour of that. Anything that can help to get people to read the information available **before** they create issues on GitHub is a good thing - for the devs and for the users.

KiteAnton, 2016-07-03:
The firmware flasher page already displays the release notes when you load fw online.

closed 2018-05-13 by stale[bot]