betaflight/betaflight-configurator #52: BF Configurator 1.4: Still no auto-update / desktop notification on update?

RipperDrone, 2016-06-26:
Under Settings / Application Options you can check "desktop notification on available updates" - this feature does not seem to work. From 1.2 to 1.3 to 1.4 I had to check for availability of new versions myself, delete the old version manually, then install the new version and re-create a taskbar button or link to start the app.

Is the auto-update feature coming soon?

mikeller, 2016-06-26:
Hi @RipperDrone.
When an update is available, try going into 'drawer / More Tools / Extensions' in Chrome and check if there is a notification showing for the Betaflight configurator. I've had this with other Chrome apps that did not automatically update due to being detected as incompatible with another Chrome extension I'd had installed.

RipperDrone, 2016-06-26:
I had checked the extensions ist in Chrome in 1.2 and 1.3 - no notification whatsoever... running Win 10 x64 on my PC.

KiteAnton, 2016-08-09:
Is this issue resolved ? Can we close this issue ?

RipperDrone, 2016-08-09:
Where's the update notification exactly supposed to show up. Until now, no sign of update hints whatsoever. HAve manually deinstalled existing version, then reinstalled updated version from Chrome store anytime an update was released. My understanding was that there was supposed to be an update notification and direct option to update instead of deinstall/reinstall, correct? Both notification and leaner update process are still not working on my Win 10 system...

mikeller, 2016-08-09:
I am not sure there is a notification that is supposed to show up after Chrome web store app updates. I sometimes get a pop up notification from chrome after an app / extension has been updated, but this does not seem to be consistent.

Seeing that nobody else is reporting this problem, this makes me suspect that there might be a problem with your installation of Chrome, or rather with the local data cached by Chrome. Can you try removing Chrome, removing all the folders named 'Chrome' under `C:\Users\<username>\AppData\...` (caution, this will remove any other personal settings / apps you have in Chrome), and then reinstalling Chrome?

RipperDrone, 2016-08-09:
I've got a lot of settings and other apps in place, so would not prefer to lose all this just to verify. I'm perfectly ok you close this thread, just wanted to let you know the status quo. It's not a major feature, and I'm aware there are more rewarding pieces to work on than this ;-)

closed 2016-08-09 by RipperDrone