betaflight/betaflight-configurator #62: PID Tab: show all PIDs function

skaman82, 2016-06-30:
This table should be hidden as well until the "Show all PIDs" button is clicked. It would be also great if the state is remembered. At the moment the PIDs are gone again on refresh.

[image]

skaman82, 2016-06-30:
@mikeller could you please take a look at this as well?

mikeller, 2016-06-30:
Done, #64. Let me know if you are good with the UI.

skaman82, 2016-06-30:
@mikeller the second part of this issue is still open

mikeller, 2016-06-30:
Is what you mean that if you click 'show all PIDs' and then refresh, only the active PIDs are showing again? If so, it's quite simple and I can do that when i get to do the rest of the work in the PID tab.

skaman82, 2016-06-30:
yes, I mean exactly that. It is a pain in the butt if you testing inflight adjustments

skaman82, 2016-06-30:
@mikeller thanks!

closed 2016-06-30 by skaman82