betaflight/betaflight-configurator #71: Can't install on ubuntu

NullVoxPopuli, 2016-07-02:
in the chrome store, after the package downloads, I get this error in a 'popup'

"Package is invalid. Details: 'Default locale is defined but default data couldn't be loaded.'."

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 15.10
Release:	15.10
Codename:	wily
```

mikeller, 2016-07-03:
Hi @NullVoxPopuli.

I think the error you are getting might be related to the version of Chrome / chromium you are using. What language are the menu texts in your Chrome / chromium?

NullVoxPopuli, 2016-07-03:
US English.
I'll check the version when I get home tonight. Good thinking!

NullVoxPopuli, 2016-07-04:
I guess restarting chrome updated it, and betaflight works now! thanks!

closed 2016-07-04 by NullVoxPopuli

mikeller, 2016-07-04:
> I guess restarting chrome updated it, and betaflight works now! thanks!

Not that unusual with Chrome. Happy to see it works now.