betaflight/betaflight-configurator #77: 3D rate model doesnt correspond to rates

AliB10, 2016-07-05:
Currently the 3D model in the PID/Rates tab doesn't move at the speed of rates on board.
I have roll rates at around 1200deg/s at full throw, and the 3D model has a roll rate of around 360deg/s

This could cause people to keep increasing their rates too high.

antstorm, 2016-07-06:
#89

borisbstyle, 2016-07-06:
Try configurator 1.6.3
https://chrome.google.com/webstore/detail/kdaghagfopacdngbohiknlhcocjccjao/publish-delayed?hl=en

closed 2016-07-06 by borisbstyle

mikeller, 2016-07-06:
@borisbstyle: Is it possible that you add tags to the repository for the releases you are publishing? This way, when users are reporting issues, we can easily verify against the exact version that they are running.

borisbstyle, 2016-07-08:
@mikeller I will add them from now. Sorry the first versions were quick releases just to get the most wanted stuff out there as quick as possible. Can settle down now

mikeller, 2016-07-09:
Good stuff. I think we're almost there. There are still a few settings that should probably be available in the GUI, and I want to add buttons for 'dump into file' and 'restore from file' to the CLI tab. Once those is done, users will be able to set up Betaflight without having to use the CLI.