betaflight/betaflight-configurator #80: Target reset requires close/restart of Configurator

MotoLab, 2016-07-05:
MOTOLAB target with Configurator on Linux consistently fails to reopen the serial port after device save/reset. Seems to be timing related, may be attempting to reopen the serial port too quickly after reset. Tested with Configurator V1.5.

MotoLab, 2016-07-10:
On further examination, this seems to be something in the firmware startup. A few seconds after the reset, Configurator will connect. I'll look into that further.

closed 2016-07-10 by MotoLab

mikeller, 2016-07-10:
I think this is caused by the 'high load on startup' issue that 3.0 is having: https://github.com/betaflight/betaflight/issues/539

MotoLab, 2016-07-10:
I think it's the same issue, but USE_SDCARD isn't defined on the MOTOLAB target so that's not the problem.