betaflight/betaflight-configurator #81: Rework 'Receiver' tab

mikeller, 2016-07-05:
Ideas that were brought up so far:

- move / add all settings pertaining to the receiver config to the receiver tab #36, #75;
- add 'check receiver settings' button #75;

haslinghuis, 2021-01-14:
Two problems while moving stuff. 

[image]

haslinghuis, 2021-01-14:
Two problems while moving stuff.
About the first the feature is filled with the RX stuff. second, don't comprehend why the i18n is not working

[image]

PR #2386 (2021-02-21): Move receiver and rssi configuration to receiver tab
Fixes: #81
closed 2021-02-21 by mikeller