betaflight/betaflight-configurator #82: Add CPU usage to taskbar (available from 3.0.0)

borisbstyle, 2016-07-05:
MSP_STATUS_EX last 16 bytes is CPU load.

Perhaps coloring would be nice

0-40 green
40-70 yellow
70-100 red

gaelj, 2016-07-16:
@borisbstyle I've just implemented this, it turns out there is a bug in the BF code:
```
    case MSP_STATUS_EX:
        headSerialReply(12);
        serialize16(cycleTime);
#ifdef USE_I2C
        serialize16(i2cGetErrorCounter());
#else
        serialize16(0);
#endif
        serialize16(sensors(SENSOR_ACC) | sensors(SENSOR_BARO) << 1 | sensors(SENSOR_MAG) << 2 | sensors(SENSOR_GPS) << 3 | sensors(SENSOR_SONAR) << 4);
        serialize32(packFlightModeFlags());
        serialize8(masterConfig.current_profile_index);
        serialize16(constrain(averageSystemLoadPercent, 0, 100));
        break;

```
headSerialReply should be set to 13.

I fixed this and will issue a PR for each.

borisbstyle, 2016-07-16:
O yeah I see. Thanks for spotting it

closed 2016-07-18 by borisbstyle