betaflight/betaflight-configurator #92: Configurator 1.6.3 - 3d model - Yaw input.

FasaArg, 2016-07-06:
The 3d model does not responde to the yaw input from the receiver.

mikeller, 2016-07-06:
What version are you using?

FasaArg, 2016-07-06:
1.6.3 , but apparently is a momentary error, after restart the configurator is working fine (strange)
Before, the signal slider worked ok, but the model does not responded.

FasaArg, 2016-07-07:
Could you reopen this issue?? The problem persist...

FasaArg, 2016-07-07:
@mikeller 
I found the problem!
If you go to PID tab after connect, works always fine.
If you go to receiver tab, does not work.
If you go back to receiver after using PIDtab, start to work ok.

mikeller, 2016-07-07:
Ha, thanks for doing the testing, @FasaArg.

@antstorm: I think this is because `currentRates` only gets initialised when the PID tab is loaded. Can you look into this?

FasaArg, 2016-07-07:
@mikeller Thanks to you, let me know if i can help with anything.

antstorm, 2016-07-07:
@mikeller sure, I'll look into this

antstorm, 2016-07-07:
Here's a PR ^. `MSP_SPECIAL_PARAMETERS` didn't get loaded, which has `RC_RATE_YAW`.

mikeller, 2016-07-07:
@antstorm: Saw that, thanks.

@FasaArg: If you are comfortable enough with GitHub to clone the betaflight-configurator repo, you can clone it, and then 'Load unpacked extension' in 'Extensions' in chrome, to verify that the fix works.

closed 2016-07-07 by mikeller

FasaArg, 2016-07-07:
@mikeller Thanks, but i'm registered since today to be honest... I have no idea.

Anyway, i will try to read about how to use it.

FasaArg, 2016-07-07:
@mikeller Today i learned how to use the clone and tested the fix, Working ok without any issue.
Thanks again.

mikeller, 2016-07-07:
@FasaArg: Thanks for testing, and well done!

PR #94 (2016-07-07): Fix model yaw input in Receiver tab. Fixes #92
Preloads `MSP_SPECIAL_PARAMETERS` in Receiver tab, which is required to calculate rotation for 3D model.

Should fix #92
commit 3cc363cb (2016-07-07): Merge pull request #94 from antstorm/fix-model-yaw-in-receiver-tab

Fix model yaw input in Receiver tab. Fixes #92