betaflight/betaflight-configurator #97: "PID loop freq" automatically changes when changing "PID math"

FasaArg, 2016-07-07:
I don´t know if this is on purpose. (Sorry if it is)
Changing from Integer to float on the PID tab, or viceversa, automatically sets PID_PROCESS_DENOM=2 and I need to go to change back on the configuration tab.
I´m running 8/8/32 on SP Racing F3 EVO but It does the same with 4K. 
I have been seeing this from configurator 1.4

---------------------------------------
2016-07-07 @ 10:22:00 -- Running - OS: Windows, Chrome: 51.0.2704.103, Configurator: 1.6.3
2016-07-07 @ 10:22:11 -- Serial port successfully opened with ID: 6
2016-07-07 @ 10:22:11 -- MultiWii API version received - 1.16.0
2016-07-07 @ 10:22:11 -- Flight controller info, identifier: BTFL, version: 2.9.0
2016-07-07 @ 10:22:11 -- Running firmware released on: Jun 25 2016 23:45:39
2016-07-07 @ 10:22:11 -- Board: SPEV, version: 0

borisbstyle, 2016-07-07:
Yes thats true indeed for f1 controllers to prevent higher cpu usage. Will
remove that in the future probably

FasaArg, 2016-07-07:
@borisbstyle Thanks.

KiteAnton, 2016-07-16:
I think this should be addressed in the Betaflight repo instead. I will re-add this as an issue there and close this one.

closed 2016-07-16 by KiteAnton

borisbstyle, 2016-07-16:
I will fix that.

It was protection to those on f1 boards who change from rewrite to luxfloat
and dont change their looptime