betaflight/betaflight-configurator #99: Move Yaw D out of the PID tuning section

etracer65, 2016-07-08:
Since Yaw D is now yaw jump prevention, I would suggest moving it out of the PID tuning section table and appropriately labeling it as "yaw jump prevention".  There continues to be confusion about this parameter relocating it and renaming it should help.  I think it should stay on the PID Tuning tab - just not in the PID values table.

mikeller, 2016-07-08:
@etracer65: Ah. I wasn't aware of this. Will move it out.

borisbstyle, 2016-07-08:
@etracer65 @mikeller 
This ain't a bug. It has been done on purpose to repurpose the D field which was unused. Very handy to not go into cli and change that.

But dont remove this yet! I will possibly re-add yaw D after I have finished my new PID algorythms for this

borisbstyle, 2016-07-08:
By the way the help tip explains it anyway

etracer65, 2016-07-08:
@borisbstyle 

I'm not saying it's a bug.  All I'm saying is that since yaw D is now actually yaw jump prevention it would make sense to move it out of the PID values table and put it somewhere else on the PID Tuning tab properly labeled and documented.  There continues to be confusion in the rcgroups forum about yaw D and the fact that it's now yaw jump prevention.

borisbstyle, 2016-07-08:
It actually has a dampenjng effect like D would have.
With the new configurator they can immediatelly see what it actually is due
to tooltip options.

etracer65, 2016-07-08:
@borisbstyle

Ok.  I understand the desire to keep it out of the CLI and why you repurposed yaw D for yaw jump prevention.  I was just suggesting a rearrangement of the PID Tuning tab to make it more obvious about what yaw D does and the fact that it's now yaw jump prevention (which is a confusing subject all by itself).  There are still people looking for yaw jump prevention in the CLI and can't find it (of course) and other people asking how to tune yaw D (which hasn't existed for a while).

@mikeller 
I'm not saying this is a bug.  More of a feature request to make the purpose of the parameter clearer.  If there's disagreement, then that's fine as it's only a suggestion.

mikeller, 2016-07-08:
@borisbstyle, @etracer65: I think we are all talking about the same, but using different ways to describe it. I labelled it as a bug because the the UI is wrong in this case: It presents a value that indicates 'yaw jump prevention' as 'yaw D'. The underlying code is right, it is just the presentation that is wrong. I agree with @etracer65 in that the value should be moved in the UI from 'yaw D' into a row of its own, to reduce the risk of users mistaking it.

PR #110 (2016-08-09): Moved yaw jump prevention to separate row. Added help text.
Fixes #99.
closed 2016-08-09 by borisbstyle