betaflight/betaflight-configurator #101: Random PID values and screen data

BoyDrone, 2016-07-09:
Hi, trying to tune my quad with a new naze32 rev6. Using betaflight 2.7 and the latest configurator. I activated super expo, and all the PIDs were displaying random values that were not set and exceeded the limit. I have a screenshot but don't have an option to upload.

Boris,
Aside from the screenis there any other data that may be useful for you before I change anything else? No rush as my batteries are all dead and require charging before trying anything again

mikeller, 2016-07-09:
Hi @BoyDrone.
Thank you for your bug report.
If the configurator showing the wrong values is still running, can you right-click into it, click on 'Inspect', and in the window that opens, select 'Console'. Then please copy / paste the contents of this window.

borisbstyle, 2016-07-09:
2.7 has no super expo.

You have to upgrade to 2.9 anyway

>
> Hi @BoyDrone.
> Thank you for your bug report.
> If the configurator showing the wrong values is still running, can you
right-click into it, click on 'Inspect', and in the window that opens,
select 'Console'. Then please copy / paste the contents of this window.

mikeller, 2016-07-10:
@borisbstyle: Agreed, but the configurator should not be showing the SuperExpo slider, and should be showing the old rates. I would like to find out what breaks this, and fix it.

borisbstyle, 2016-07-10:
Yes I agree about that.

But just wanted to say to him that he needs 2.9 to be able to enable it
anyway ;)

mikeller, 2016-07-11:
Had a quick check, and for firmware < 2.8.0, the 'SuperExpo' slider should not be showing.

@BoyDrone, can you please check again and confirm that this is still broken for you?

BoyDrone, 2016-07-12:
@mikeller as of last night, I upgraded to 2.9 to see if that had any affect. from what I can tell, the betaflight configurator shows values in different units based off of what cleanflight diplays.

ie. if cleanflight shows I = 0.200, betaflight displays I = 200 and so on, this was the case for Ps & Ds as well. but the factor differred between them all. I'll have to take some screenshots tonight to evaluate and share.

the biggest issue I had with this data sorting was that once I plugged in my OTG + USB cable to my phone and used EZGUI to adjust PIDs, the board wouldnt accept the values since the max for P is 20, and betaflight expects x10 values... so setting a number to 5 would mean you'd have to input 50, which the system doesnt accept.

I hope this makes sense (just checking emails in between meetings at work!)

mikeller, 2016-07-12:
Hi @BoyDrone.

> ie. if cleanflight shows I = 0.200, betaflight displays I = 200 and so on, this was the case for Ps & Ds as well. but the factor differred between them all. I'll have to take some screenshots tonight to evaluate and share.

This is to be expected. The PID values in the Betaflight configurator are scaled up to make them appear identical to the values you get when using the CLI. Also, be aware that all settings (including PID values) got reset when you upgraded to 2.9.0, so this would explain why they look different to what you had before.

> the biggest issue I had with this data sorting was that once I plugged in my OTG + USB cable to my phone and used EZGUI to adjust PIDs, the board wouldnt accept the values since the max for P is 20, [...]

See above, the values that EZ-Gui expects are the not - scaled - up values that cleanflight configurator uses as well, and not the scaled up ones.

I am still interested in fixing the bug that causes you to see the 'SuperExpo' slider when using Betaflight 2.7. Can you go back to the version you had when you got the problem, and report on any errors that you are getting in the console?

@borisbstyle: I don't know, the scaled up values seem to be confusing. Should we add a help text somewhere, or should we go back to the unscaled ones to be in line with other tools?

borisbstyle, 2016-07-12:
I liked them like this. No need to lie about the real pids.

The betaflight configurator shows the real values like you see them in your
dump.

I think it is confusing that cleanflight tools show different values in
configurator and cli

mikeller, 2016-07-12:
> I liked them like this. No need to lie about the real pids.

Ah ok. I always assumed the values in the cleanflight configurator GUI where the real ones, and they were just upscaled in the CLI and firmware to make them into ints. If the int values are the actual coefficients then I really don't see why the cleanflight configurator would scale them down.

I will look into adding a help text to tell people that they will have to convert the PIDs if they want to use them in EZ-Gui and suchlike.

BoyDrone, 2016-07-13:
@mikeller Sorry, I had to rebuild my quad last night. complete tare down and rebuild... the Rev6 board is remvoed, but I can still program it and test it this evening. I should definitely have some time to do this tonight to verify with screenshots today.

@mikeller & @borisbstyle 
regarding the scaled-up values, or help note:
Why not just leave the non-scaled values as they are and just have a slider/toggle button up top somewhere to switch between scaled/non-scaled values. I agree that there is no sense in scaling the values if the CLI sees them as a base unit of x, but considering the various configurators out there and people switching between configurators, some may find this feature of toggling between one and the other appealing to stick with betaflight configurator or start using it more frequently from their norm.

BoyDrone, 2016-07-14:
Hi guys, ok. finally just got a chance to take a couple screenshots.
Let me know if this helps?

[image]

mikeller, 2016-08-01:
Closing as no further info.

closed 2016-08-01 by mikeller

BoyDrone, 2016-08-03:
bah. i finally got to this today. you just wanted to see super expo? its there in 2.7 on rev6

mikeller, 2016-08-03:
Ah, now I think I understand. Is the `SUPER_EXPO` that you are seeing the one on the 'Modes' tab, like in your screenshot? If so, this is where it should be for 2.7. It has been moved in the PID tab for later versions of the firmware, and @borisbstyle and I assumed that was where you were seeing it.

Do you still get the random PID values when you are using the latest configurator (should be updated automatically) with 2.7?

BoyDrone, 2016-08-05:
Hi Michael, I had a rev5 board fry on me a couple days ago. User error... not the board. oops.
so I reverted back to to Rev6, and installed 2.9.1 on it. everything seems to be corrected from what I saw on 2.7. Do you want me to go back to 2.7 tonight and plug in EZGUI to see if the PID values keep changing between device plug ins?

mikeller, 2016-08-06:
Hi @BoyDrone . Thanks for offering, but there's been a number of changes in the configurator since you've reported this, so the issue you had was most likely fixed. If not, somebody else will report it, without the need for you to flash an older version.