betaflight/betaflight-configurator #104: Throttle mid setting is mapped to the wrong variable

rav-rav, 2016-07-11:
Throttle mid setting sets  'midrc' instead of 'thr_mid'.
While thr_mid is used for throttle expo, midrc is the middle point for roll/pitch/yaw.

mikeller, 2016-07-11:
Hi @rav-rav.

Where do you see this? I believe they are mapped correctly in the PID tuning tab:

https://github.com/betaflight/betaflight-configurator/blob/master/tabs/pid_tuning.js#L196

https://github.com/betaflight/betaflight-configurator/blob/master/tabs/pid_tuning.js#L292

rav-rav, 2016-07-12:
I meant this setting in the configuration tab:
https://github.com/betaflight/betaflight-configurator/blob/master/tabs/configuration.js#L509
https://github.com/betaflight/betaflight-configurator/blob/master/tabs/configuration.js#L603

It says "Middle Throttle" but has actually nothing to do with throttle. The underlying command 'midrc' is used to set the middle point for your TX/RX values. If you change this you will always have a drift in all received commands, like trim. It's there for TX/RX combos not having 1500 as center value.
I believe the place where this command can be set and its name are misleading.

mikeller, 2016-07-12:
Ah yes. Looks like the label on this one is wrong. Will fix.

@borisbstyle: Do the `minthrottle` and `maxthrottle` values above / below this apply to all RX channels as well?

borisbstyle, 2016-07-12:
Yes....mid_rc to all channels

mikeller, 2016-07-12:
`minthrottle` and `maxthrottle` (by name in MSP) are for throttle only, or are they for all channels as well?

borisbstyle, 2016-07-12:
Minthrottle and maxthrottle is output to the esc

PR #109 (2016-07-12): Fixed label for midrc. Added some help texts.
Fixes #104.
closed 2016-07-12 by borisbstyle