betaflight/betaflight-configurator #115: BF 2.9: smartport telemetry occasionally not working

nikybiasion, 2016-07-16:
If i set gyro update and pid  loop at the same frequency, smart port telemetry stop working.
i've tried 2k/2k 4k/4k 8k/8k
If i set 4k/2k 2k/1k 8k/4k smartport telemetry works

I use a spracing f3 evo with this port configuration:

serial 20 1 115200 57600 0 115200
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 32 57600 57600 57600 57600

KiteAnton, 2016-07-16:
I think you should post this in the issue section of betaflight instead,  this repo is only for the configurator.  
Please also add some additional info regarding tasks and status for the settings you tried.

borisbstyle, 2016-07-18:
This is not configurator issue

closed 2016-07-18 by borisbstyle