betaflight/betaflight-configurator #122: LED: Overlay choices stay on selected LEDs when choosing 'Ring'

mikeller, 2016-07-18:
In the LED config tab, when one or more LEDs are selected, there are a number of overlays to be selected from when 'Function' is set to 'Color', 'Modes & Orientation', or 'Arm State'. When 'Ring' is chosen as 'Function', the overlay section is hidden, which makes sense.
However, if overlays are active for the currently selected LEDs, these are not cleared, which is a bit confusing.

I think we should either:
- not hide the overlay selection when 'Ring' is selected
- clear the overlays for the currently selected LEDs when 'Ring' is selected.

Apart from that good work on the new LED code, it looks nice.

gaelj, 2016-07-18:
@mikeller @borisbstyle 
- overlays are not used by the controler, when in Ring mode. So the configurator does not show them. The idea here is to only show options that are acutally available.
- unavailable overlays are cleared when changing mode via the drop-down box
- the bug you mention only occurs when such an "invalid" configuration was previously set. This can happen in two cases: either from the command line, or inherited from a restore. (right ?)

I had considered this case and decided to not modify anything in the configurator.
The idea being:
- always display what is actually memorised in the controler, even if this is invalid
- prevent invalid settings by hiding UI elements
- refuse invalid configurations in the BF LED code - this prevents invalid settings regardless if they are set by CLI or MSP. This is part of https://github.com/betaflight/betaflight/pull/737 (which was reverted).

mikeller, 2016-07-18:
Ah, so once https://github.com/betaflight/betaflight/pull/737 has been merged, the invalid overlays will be removed once the config is saved?

I think this is a good way of doing it, except I'd maybe show a notice like 'preexisting overlays set, will be deleted when saving' instead of the overlay selection if there are overlays and 'Ring' is selected.

gaelj, 2016-07-18:
Not exactly. 737 will prevent any invalid settings from being applied. If the user tries to apply an invalid setting, the old is setting will remain unchanged.
This cannot happen in the GUI of BFC as the elements are hidden. It is only possible in CLI mode or when restoring, in which case displaying a notice will be more difficult (BF is refusing the configuration, BFC is unaware if it is valid or not).

gaelj, 2016-07-18:
I could modify BFC so that, upon displaying the LED tab, it will check for invalid settings and correct them, but this will only apply to current BF versions. Is it worth the effort ?

mikeller, 2016-07-18:
I think what you want to do is the right thing. But after your description, I think there is a bug in the configurator LED code. This is how my actual config looks after selecting LEDs 3, 2, and 1:

[image]

If I then change the function to 'Ring', the 'Warnings' and 'Throttle' overlays on LED 3 stay on, despite this not being a valid config according to what you say:

[image]

gaelj, 2016-07-18:
OK, thanks for spotting this. I will correct.

gaelj, 2016-07-18:
@mikeller problem fixed. Any comments welcome.

closed 2016-07-18 by mikeller

mikeller, 2016-07-19:
@gaelj: Sorry to rain on your parade, I think I've found another issue with multiple selections:

When I first open the LED tab and select 3 LEDs, everything looks ok. But when I select multiple LEDs with different functions, the modifiers are still available on the right hand side:

[image]

This enables me to enable modifiers for LEDs with functions that should not support them (the very left one is 'Battery' in the example):

[image]

Also, these 'illegal' choices seem to get persisted on 'Save'. This is after Save / FC reset / configurator restart:

[image]

skaman82, 2016-07-19:
@mikeller Wow I never thought that multiple selection was possible at all! How do you did that? Tried ALT or Ctrl keys but I still can only select just one LED.

mikeller, 2016-07-19:
:-)

gaelj, 2016-07-19:
@mikeller fixed

PR #126 (2016-07-18): Don't display first invalid LED / Fix overlay not disabled properly
- Don't display first invalid LED (defined as function = color, color = 0, no directions, no overlays, position = 0:0)
- Fix: unavailable overlays are not always disabled when changing function on multiple selection of LEDs (fixes #122)