betaflight/betaflight-configurator #123: The PID Tuning page shows transposed Horizon Strength & Transition values compared to Cleanflight

TonyLeyland, 2016-07-18:
It seems the Horizon values "Strength" and "Transition" are transposed on the PID Tuning page. 
I double checked by firing up the latest CleanFlight Configurator on the same copter.

I know I have a strength value of 6 and a Transition value of 50 and Betaflight Configurator shows these values in opposite places to Cleanflight Configurator. 

I've attached the 2 GUI screens here for reference.

Thanks
Tony

[image]

[image]

mikeller, 2016-07-18:
Fixed in #100. Try updating to the latest version of the betaflight configurator.

TonyLeyland, 2016-07-18:
Cool, thanks for letting me know. Looks like 1.6.4 was released only a short time ago. 

Everything looks great, cheers.

closed 2016-07-18 by mikeller