betaflight/betaflight-configurator #127: Documentation / LED strip

19Gerhard85, 2016-07-19:
Hey guys,

is there any documentation about the new LED strip features? e.g. which functions does what, what are the priorities, etc.?

For example when I set the larson scanner and the arme state to several LEDs, which function will have more priority to be displayed?

Greets

skaman82, 2016-07-19:
Armed state would be the priority function. Larson scanner would be an overlay effect on top.

gaelj, 2016-07-19:
Hi @19Gerhard85 
Updated documentation can be found in my CF PR, which is not yet merged, so in my fork for the time being:
https://github.com/gaelj/cleanflight/blob/LED-battery/docs/LedStrip.md

Basically to each LED is assigned a function, which will define its "basic background color".
Additionnally and optionally, you can activate some color modifiers, blink functions, and overlays to this basic color.
When using the configurator GUI, you will only be presented with viable combination possiblities - previously everything was possible, including many non-working combinations.
Note that for Battery function you need to activate the current sensor (see CF documentation about the virtual current sensor if you don't have a physical one - it works great).
Larson scanner needs to be selected on several LEDs to have the desired effect (I'd recommend at least 4).
In your example the color of your LEDs will show armed state, and the larson modifier will change each LED's brightness in an animation way.

19Gerhard85, 2016-07-20:
@gaelj 
Awesome, thank you! Is it now possible to change the armed / disarmed color?

closed 2016-08-14 by mikeller