betaflight/betaflight-configurator #129: No scroll bar on left-hand frame - can't access CLI on laptop screen

joshuabardwell, 2016-07-19:
On my laptop, I can't access the CLI because the left-hand pane does not have a scroll bar. On the Cleanflight Configurator, there is a scroll bar that lets me access them.

[image]

[image]

deepimpactio, 2016-07-19:
Use the force Jedi ;)

borisbstyle, 2016-07-19:
You seem to have some issues wirmth graphics too as the 3D model doesnt show up. I have seen this issue with someone else too who had 2 video cards in his pc activated.

KiteAnton, 2016-07-20:
I have the same problem when using an old laptop with not enough resolution. Since adding the OSD tab the CLI tab got below what would fit the screen. If you resize the entire betaflight window then the scrollbar will appear.

joshuabardwell, 2016-07-22:
The 3D model is not showing up because I am using a custom mixer. Completely normal.

KiteAnton, 2016-07-22:
@joshuabardwell I was more refering to my own laptop (it is a cheap 11" one from 2011) not yours :) 

I agree, resizing is not a solution just a workaround. It was more to give some tips to anyone trying to find the CLI until the bug is fixed. 

I will investigate this if I get some free time during the weekend now.

skaman82, 2016-07-22:
Fixed in https://github.com/betaflight/betaflight-configurator/commit/c3b4cb82f30045b345d9459bc36acd65c4594a09

closed 2016-07-22 by skaman82