betaflight/betaflight-configurator #141: Feature Request for disabling unneeded tabs

TheAngularity, 2016-07-22:
would it be possible to disable some tabs if they have not been activated this feature in the configuration tab? The configurator would become clearer and you configure only what's needed.
This concerns gps, osd, transponder, led strip and blackbox.

skaman82, 2016-07-22:
It would be nice if hidden tabs would show up automaticly on on feature activation (GPS, LED, Transponder and so on)

KiteAnton, 2016-07-22:
Yes I think could make sense, as it is now there is limited space available for some laptops with small screens so not all tabs fit properly.

borisbstyle, 2016-07-22:
Yes great idea!

TheAngularity, 2016-07-22:
> - OSD Tab (if the hardware supports it)
> - LED, Blackbox, Transponder, GPS (if the feature is activated in the config tab)

And for Adjustments, Servos and Tethered Logging there is also determined a solution.

nathantsoi, 2016-07-23:
This is a great idea :)

mikeller, 2016-07-23:
Anybody already working on this?

TheAngularity, 2016-07-23:
> Anybody already working on this?

I just had the idea, so no (insufficient programming knowledge) ;)

mikeller, 2016-07-23:
All good, I'm happy to do it, I'm just trying to avoid doubling up on work that's already being done. And thanks for making the feature request,  @TheAngularity.

TheAngularity, 2016-07-23:
@mikeller 

Thats nice, thx. I only make a quick & dirty solution for my self.

I'm insert this in serial_backend.js from line 266

```
	$('#tabs ul.mode-connected li.tab_servos').hide();
	$('#tabs ul.mode-connected li.tab_gps').hide();
	$('#tabs ul.mode-connected li.tab_osd').hide();
	$('#tabs ul.mode-connected li.tab_transponder').hide();
	$('#tabs ul.mode-connected li.tab_logging').hide();
```

TheAngularity, 2016-07-25:
closed now ... see #142

closed 2016-07-25 by TheAngularity