betaflight/betaflight-configurator #146: Receiver tab is crashing

skaman82, 2016-07-26:
[image]

mikeller, 2016-07-26:
Tried to reproduce this, but receiver tab loads fine for me in current master. Did you get any log info in the console? How did you get there?

skaman82, 2016-07-26:
@mikeller  I just clicked on the Receiver tab. Console showing this:
[image]

mikeller, 2016-07-26:
Could be related to the changes I made in the firmware. See #147.

closed 2016-07-26 by skaman82