betaflight/betaflight-configurator #147: PID type is hidden in the PID Tab

skaman82, 2016-07-26:
[image]

mikeller, 2016-07-26:
Tried to reproduce this, but PID selection worked fine for me in master. Did you get an exception or other logs in the console? What steps did you take to get there?

skaman82, 2016-07-26:
No errors in the console. But timeouts.
[image]

mikeller, 2016-07-26:
Ha, could be related to the changes I made in https://github.com/betaflight/betaflight/pull/821. Have you tried updating your firmware?

skaman82, 2016-07-26:
@mikeller will try that. I can compile the newest master at home (9 om and still at work). I am running a version from  3 Jul on my test board here.

borisbstyle, 2016-07-26:
Probably your version is too old. Not worth trying to solve these things.
The pid controller order for example changed.

All unreleased versions dont count in terms of support.
3.0 is still not finished and more will be added. It is a huge one and
thats why it takes longer.
Before looking into issue always verify against the latest master.

mikeller, 2016-07-26:
@skaman82: You should be able do download the latest build from http://andwho.sytes.net:8080/job/BorisB_BetaFlight/lastSuccessfulBuild/artifact/obj/

closed 2016-07-26 by skaman82