betaflight/betaflight-configurator #148: PID Tab clean up

skaman82, 2016-07-26:
@mikeller I would like to do a cleanup of the GUI elements. For that I need the following graph to be 50px shorter in height.  I could not figure how to get this done. Could you do this? Then I can start my cleanup.

[image]

mikeller, 2016-07-26:
Will do, @skaman82. There is a couple more settings to go into the pid tab once @borisbstyle has added the enhancements to the firmware, so it's probably best if you wait with the redesign until all the elements are on the page.

skaman82, 2016-07-26:
@mikeller I intent just to clean up an fix the GUI stuff, not completely redesign it. The are some Interface Elements broken or out of place.

mikeller, 2016-07-26:
I know. I probably added most of them, and then didn't know how to make them look good. ;-) I'm just saying there is more stuff coming that will need cleanup and probably rearrangement, and I am a bit wary of having to merge the profile / rateprofile related stuff that I am working on at the moment with other html changes, as html does not merge well.

skaman82, 2016-07-26:
@mikeller After your last commit the PID tab ist broken on my side. Configurator freezes too.

borisbstyle, 2016-07-26:
@mikeller
Code is slowly coming into place. I solved some issues I had and had
successfull flights today.

So basically the thing I mentioned about 2 sliders is still on where I
earlier said it needs to be only one.

What MSP command should I use?

I also want to add rc smoothing as well with a drop down menu as well.

mikeller, 2016-07-26:
Since we already got `deltaMethod` in `ADVANCED_TUNING`, it probably makes sense to stick it in there. I assume that `deltaMethod` will only apply to the legacy PID controller, so you might as well reuse this one, and rename it to something less specific like `deltaParameter`.

skaman82, 2016-07-26:
@mikeller just tested with latest firmware master. Receiver tab is ok now. PID Tab still broken and freezes the whole app.

[image]

mikeller, 2016-07-26:
F*ck. I made it worse with the 'fix' I deployed. Will fix after breakfast.

skaman82, 2016-07-26:
@mikeller fixed via https://github.com/betaflight/betaflight-configurator/commit/ed07aeaa2dd34baf54058c1c9afba57cb1096a6b

mikeller, 2016-07-26:
Thanks, @skaman82.

skaman82, 2016-07-31:
@mikeller also took care if this one in https://github.com/betaflight/betaflight-configurator/commit/e97f6685d43d9899d36f42dcb771bf183ff40e88

[image]

closed 2016-07-31 by skaman82

TheAngularity, 2016-07-31:
@skaman82 
like it ... but what mean's the two SUPEREXPO_RATES an Super Expo Rates ... i think the second is the real one ... hmmm, and where is the tooltip on Super Expo Rates

skaman82, 2016-07-31:
@TheAngularity I just cleaned up the layout. I din't add or removed stuffyet. But I get what you mean. I will remove the second label SuperExpo. The tooltip is not up to me but I can put a placeholder in there.

skaman82, 2016-07-31:
@mikeller I also think we should remove the marked table cell. I just took a whirl at I but I didn't get where to remove it in the JS file :(

[image]

mikeller, 2016-08-01:
@skaman82 All the switches for features that are spread across the configuration / failsafe / PID tuning tabs are generated by the same code, and include the switch, the feature name (`SUPEREXPO_RATES`), and a description. I would like to keep it this way, and keep them generated, as this will allow us to add more features and have them show up in the right place automatically.

But I totally agree that some of the descriptions for features are not ideal for this use case and should be reworked.