betaflight/betaflight-configurator #149: Todo: add band and channel selection an power setting in OSD Tab (SIRIN FPV target)

skaman82, 2016-07-26:
At the moment user can select only channel 0 - 39 in the osd tab. This is not convenient at all.

We should add two select fields:
1. Band (Boscam A, Boscam B and so on)
2. Channel (should change depending on selected band and show CH1: XXXXMHz)

It would be also nice to set power setting "vtx_power" in the osd tab.

skaman82, 2016-07-26:
@savaga have you thought about the possibility to add a custom nickname and OSD_switch functionality to hide the osd or display a different layout setting?

skaman82, 2016-08-01:
[image]

Tomcatt, 2016-08-16:
Why if I didn't know any better, there would be no need to use the OSD Configurator anymore.  Turnkey

One less program to install.

lazd, 2016-09-13:
@skaman82 I don't think it makes sense to do this on the OSD tab, as this is technically a separate thing. It may make sense to add a VTX tab, though it feels like a waste since there are only 3 options.

I'm going to investigate exposing UI for this and will send a PR.

PS: @skaman82 the discussion regarding OSD switch etc should be in a separate issue.

skaman82, 2016-09-13:
@lazd I don't se why we should do a separate tab for this - as you say, this would be a waste of space. There must be some kind of target detection in the OSD tab though to show the vtx setting only if supported by connected target. We also could do this in a sub-tab (like in the PID Tab).

lazd, 2016-09-13:
@skaman82 the VTX is not the OSD, so it should not be on the OSD tab, either as mocked up in your screenshot or as a sub-tab.

It is not a waste of space, it surfaces the information in an obvious place. I am focusing on doing this work in a separate tab right now and will defer to the project owners as to whether this is the right place for it.

skaman82, 2016-09-13:
@lazd ok, if you need some help with graphics/css or the VTX tab icon, let me know.

lazd, 2016-09-13:
@skaman82 a VTX icon would be nice, though I have no idea what it would look like!

skaman82, 2016-09-13:
@lazd added icons in https://github.com/betaflight/betaflight-configurator/commit/ffb6cfe0810b2811bc827129773bffd3ba976501

[image]

lazd, 2016-09-21:
@skaman82 Alright, I'm a bit stuck. I tried doing `MSP.send_message(MSPCodes.MSP_VTX_CONFIG, false, false, load_html);`, but I get no data back and the `FC reports unsupported message error: 88` message appears in the console.

How can I go about fetching the `vtx_channel` and `vtx_power` from the VTX, updating it, and writing the value back?

mikeller, 2016-09-21:
@lazd: Are you using a singularity FC? That is the only board that supports VTX at the moment.

lazd, 2016-09-21:
@mikeller actually I'm using the SirinFC, which also has a VTX and OSD integration with Betaflight. It relies on the `vtx_channel` and `vtx_power` CLI commands to change the channel, and also allows configuration via the OSD.

I could not find documentation on how to do something like "change this CLI setting" via a tab in the configurator, and I gather it needs to be done with `MSP.send_message`, but I guess I've got the wrong code...

Is there something that needs to be added on the FC side?

mikeller, 2016-09-21:
@lazd: Currently only the SINGULARITY target in Betaflight supports VTX and `MSP_VTX_CONFIG`. This will probably be added to the SIRINFPV target in the future.

lazd, 2016-09-21:
@savaga can you look into adding support for `MSP_VTX_CONFIG` to the Sirin target?

savaga, 2016-09-22:
@lazd, I will take a look and send you a test build

lazd, 2016-09-22:
@savaga excellent! Also please see if `MSP_SET_VTX_CONFIG` is something we need to store the values. Thanks!

closed 2024-05-06 by haslinghuis