betaflight/betaflight-configurator #152: Failsafe stage2

skaman82, 2016-07-26:
I can not enable stage2 in the failsafe settings. If I enable stage2 and click save nothing happens.  If I want to go to another tab after this, the configurator hangs. Following console error on save in the failsafe tab. Tested with lasest configurator/firmware master.

[image]

mikeller, 2016-07-26:
Oops, didn't check that one when I switched over the features code to be more self contained. Will look into it.

skaman82, 2016-07-31:
@mikeller just fixed the layout via https://github.com/betaflight/betaflight-configurator/commit/37164a2bbaccfcb344c3b88f0f60735a6fb3903b I think we can close this now

mikeller, 2016-07-31:
Ha, you are a CSS wizard. Thanks!

mikeller, 2016-07-31:
By the way, if you add 'Fixes #x' as a comment to your pull requests or commits, github will close ticket x automatically.

closed 2016-07-31 by mikeller