betaflight/betaflight-configurator #163: External minimOSD support in configurator

llambkin, 2016-07-31:
Hey guys,

I got excited about the new OSD tab that is in the configurator now. But I have since found out it is for specific boards only.

I was wondering if support for existing minims can be integrated into the Betaflight configurator? It would be awesome if it was a one stop shop for the entire copter.

At worst case, having a button on the OSD page the allows passthrough to MWOSD configurator would be a good idea. I currently have 2 boards with a VCP and an OSD on a spare UART, and the minim is buried in the craft. Being able to configure it without having to disassemble it would be great.

mikeller, 2016-07-31:
At least for the 'not having to disassemble' part, isn't this taken care of by the 'serial passthrough' functionality?

But yeah, having the OSD tab support serially connected OSDs would take this to the next level. :-)

llambkin, 2016-07-31:
It does? If this functionality is available, it isnt well documented. I had no idea that I could do this. Maybe it could be added to the documentation....

But yeah, having OSD configuration in the GUI would give our project a polish that no other project has. It would require work from people far smarter than me. Its about time we had our own OSD, and call it BFOSD. Its about time we got rid of the last Multiwii code yeah? I wonder if someone could make a fork of Shiki's code to work with our GUI saving having to write the code from scratch.

RBeaubien, 2016-08-02:
Are there any specifications for designing these specific OSD boards?  I'm integrating OSD into a project of my own and would like to use this facility.

Tomcatt, 2016-08-16:
If you want to see what it looks like check out this posting on #218

closed 2018-05-13 by stale[bot]