betaflight/betaflight-configurator #169: Remove this thing from google chrome

averter, 2016-08-02:
I couldn't believe when I saw this. It is inconceivable that I have to install a different and so insecure/non-privacy concerned web browser as google chrome to configure my quadcopter. For me this is not how an open source project should work. I consider this a major flaw and mark it as a high priority issue. Make betaflight-configurator a standalone application.

KiteAnton, 2016-08-02:
Both cleanflight and prior to that baseflight uses this kind of platform for configuration. One benefit with this approach is it makes the development platform independent. Until we find a better suitable platform-independent solution I think this is how it is going to be for a while. 

Do you have any better suggestions? Anyone are able to make PR to this repository and suggest changes so if you have something going good already then please share your work/ideas.

KuchenKerze, 2016-08-02:
You don't have to use the Chrome App. Feel free to use the cli in your terminal of choice: https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md

mikeller, 2016-08-02:
@averter : Just out of interest: Do you have any suggestions on what framework could be used to implement a multiplatform GUI application (like the configurator), that doesn't have the drawbacks you describe above?

averter, 2016-08-02:
@KuchenKerze : Great alternative and thank you for pointing this out. For the time being I suggest that this option should be in the installation section of the readme file. 
I am new to betaflight so for now I can't comment on what would be the ideal framework for a multiplatform GUI application. Eventually I will share my thoughts on this.

KiteAnton, 2016-08-02:
It is always good with new ideas and suggestions. However since we don't have any solution for this I will close this issue. 

Anyone have any good suggestions/recommendation for multiplatform framework are encouraged to propose their ideas.

closed 2016-08-02 by KiteAnton

mkroehnert, 2016-08-02:
You could take a look at Electron (http://electron.atom.io/) which is used as the base for the Atom editor.
This is probably the closest to the current Chrome application.

mikeller, 2016-08-02:
Electron sounds promising indeed.

But: '...electron uses Chromium and node.js...'. In this respect it does not address the concerns of the original poster, as the current app already runs just fine under Chromium.

mkroehnert, 2016-08-02:
@mikeller there is a difference between Chromium and Chrome.
Chromium is the open-source part used by the proprietary Chrome browser.

While privacy concerns about Chrome are understandable I would call neither of them insecure.

averter, 2016-08-02:
@mkroehnert Even better then. The best solution (at least for me) would be to use Chromium by default instead of Chrome. This would solve the privacy issues and ensure the security of the data generated by the usage of betaflight. The software works fine on both so... 

Note: I don't want to escalate this conversation into a privacy related issue. I am just pointing out a concern which I personally have.

skaman82, 2016-08-02:
I am just wondering about this. I can't understand why it is a problem to use chrome? Never get why some people have privacy concerns. If Google wants to know if I use betaflight, so what... I couldnt care less. But that is my personal opinion.

averter, 2016-08-02:
@skaman82 Yes at the end of the day it all boils down to personal opinion. But if you claim that a software is opensource you should have a software which you control, all of your data, the software, who you share it with, the statistics, everything. 
Users can always use chrome but the default I think should be open source based like chromium (my opinion).

mikeller, 2016-08-02:
@mkroehnert: I am well aware of this. But I am not sure if the original poster was, or else why would he have taken objection with the Betaflight configurator being a Chrome app, and thusly being able to run in chrome and chromium.

@averter:
> Users can always use chrome but the default I think should be open source based like chromium (my opinion).

What do you mean by 'default'? It is not like the Betaflight configurator defaults to running in one or the other, it happily runs in whichever of the two browsers it happens to be installed in. If you do not feel comfortable with installing the app from the chrome web store (which is the default way of installing apps for both browsers), you can always clone it directly from github and use Menu / Extensions / Install unpacked to install it.

averter, 2016-08-02:
@mikeller Precisely. I wasn't aware of chromium existence until mkroehnert announced it. And just like me many others might not know. 
By default I meant to simply suggest Chromium instead of Chrome in the installation section of the software.

mikeller, 2016-08-02:
> By default I meant to simply suggest Chromium instead of Chrome in the installation section of the software.

Good point. we should at least mention chromium in there. Can you please point out where in the documentation you found this?

Also, if privacy and openness are a concern for you, may I suggest that you try an alternative operating system like linux? Most linux distributions come with the chromium browser by default, instead of chrome.

averter, 2016-08-02:
Thanks @mikeller I think I'm not the best person to know where this information should be written. I haven't even seen the documentation. The installation section of the readme file of the source code was my only reference.
I've always used linux  :-)  Just didn't ever came across Chromium (used to firefox).

mkroehnert, 2016-08-02:
@averter great to hear that I could make you aware of Chromium.

@mikeller from your comment I wasn't entirely sure if the difference was known. But now, everything seems to be settled.

Before adding Chromium to the documentation you should be aware that Chromium does not seem to provide official packages for download.
So the only way to easily install it seem to be Linux package managers.