betaflight/betaflight-configurator #192: Yaw RC rate missing from adjustments

Yamaford, 2016-08-08:
Was just about to set up my rates to inflight tune and noticed yaw rc not there.
Thanks

KiteAnton, 2016-08-08:
Yaw RC rate is currently not supported in Betaflight FW, see available functions [here](https://github.com/betaflight/betaflight/blob/master/src/main/io/rc_controls.h#L177-L201). 

Until Yaw RC rate is added to betaflight fw there is really not much that can be done in the configurator. I will close this issue, please add an issue/PR to [betaflight repository](https://github.com/betaflight/betaflight-configurator) if you really need this feature and then this issue can be re-opened after it has been added to the FW.

borisbstyle, 2016-08-08:
it can be added

KiteAnton, 2016-08-08:
Ok I added a PR for this to betaflight, I can add it also for configurator when the change is merged.

KiteAnton, 2016-08-09:
This is fixed by #195

closed 2016-08-09 by KiteAnton