betaflight/betaflight-configurator #199: Save button on Blackbox Tab is broken

TheAngularity, 2016-08-09:
I can change quite normal but unfortunately not save on tick the button :(

KiteAnton, 2016-08-09:
Which version of configurator and betaflight firmware do you experience this with?

TheAngularity, 2016-08-09:
Configurator 1.7.2 from last commit and BF 3.0.0 also last commit

KiteAnton, 2016-08-09:
Thanks,  I will try to check this later. Do you get any error/warnings in the inspector console?

TheAngularity, 2016-08-09:
cant see any error or warning

KiteAnton, 2016-08-09:
@TheAngularity I can't reproduce your problem. Save button works for me on latest configurator from github. I actually doesn't reboot the board as it states but data is saved.

TheAngularity, 2016-08-09:
here is my way to reproduce the problem:
- go to blackbox tab ... change something and save ... it's save but not reboot
- go to cli ... then type exit to reboot
- back to blackbox tab ... and the old values befor changed are back

KiteAnton, 2016-08-09:
Yes ok, I can reproduce the same. Will investigate this, maybe the missing reboot will fix the issue.

KiteAnton, 2016-08-09:
Ok, found the issue and fixed it. @TheAngularity thanks for reporting this, I was not aware it didn't work.

PR #201 (2016-08-09): Fix save issue for blackbox tab
Fixes #199
closed 2016-08-09 by KiteAnton