betaflight/betaflight-configurator #208: Modify interface to fully display on small screens like 8" win10 tablets

michelp29, 2016-08-12:
The default interface does not display correctly on my 8" win10 tablets (Chuwi Hi8 pro with original windows 10, 1920*1200 screen)
- the "save and reboot" button does not displays, even when scrolling completly to the bottom of the window, making it unuseable.
- the left vertical icon bar does not fully display, even when scrolling completly to the bottom, making CLI not accessible in horizontal mode (but can be accessed in vertical mode)
- generally, the bottom infos can not completely be accessed du to a bug with scrollers.

The problem appears both in MS win10 settings : "tablet mode" or "normal/desktop mode".

In normal/desktop mode, it is possible to switch the BFC windows to some sort of "not full screen", so it appears larger (!), and to drag it partially off screen until a part of the "save and reboot" appears, making it clickable.

I'm using it on the field for my FCs and not very happy with this problem, dispite BF+BFC are a must-have ;-) Hope this will be corrected, and thanks for the work.

[image]
[image]
[image]
[image]

mikeller, 2016-08-13:
Hi @michelp29.

This is a hard one. It is very hard to get display on mobile devices right without having access to the device for testing, and even then it requires a lot of work, as it needs to be tested for a lot of different devices.

In general, the configurator is designed to work in a window, and not in full screen. Once switched to be in a window, you should be able to resize the window so that it fits onto your screen, and then the controls should display at the bottom of the window.

Can you please try to see if this works on your device?

michelpozio, 2016-08-16:
Hi @mikeller ,

Unfortunately, I can't correctly resize it in window mode : the minimum size of the app window is larger than the display viewport of the tablet, despite the app minimum size is something like 1041x589 only (and this size works well on a desktop screen).

It seems to be due to the retina screen (283dpi, 190x1200) or a DPI/digital-point-per-inch problem or a pixel ratio thing, as the app window seems to be scalled up on the tablet. May be it's a windows 10 conf problem, I don't know...

mikeller, 2016-08-16:
Unfortunately, I don't think there is much we can do for this at the moment. With all the different screen types and geometries, supporting mobile devices is never going to be a 'one size fits all' thing, which is all we can really do in the limited time we've got.

michelp29, 2016-08-17:
I finally found a workaround by modifying display defaults in Windows 10 : 

For anybody who will encounter the same display problem,
just go to : Settings > System > Display, then  modify the cursor named "Change the size of text, apps and other items" to whatever works for you. 
In my case it was "200%" (default) and works just fine when changing it to 150%.

Scrollers are ok, all buttons and bottom-page infos can be accessed, and it even works in fullscreen or/and in tablet mode now :-) (but in landscape only) 

So it seems to be just a windows conf issue, sorry for the "bug" filling.

mikeller, 2016-08-17:
Good to hear that you found a way, and thanks for sharing so that others can benefit too.

skaman82, 2016-08-18:
I think you had some settings messed up on your device. I just tested the current chome webstore version on my 8" win10 tablet and it works just fine in the landscape mode like it always have.

[image]
[image]
[image]
[image]
[image]

skaman82, 2016-08-18:
The minimum window size of the app is 1024x550px by the way

skaman82, 2016-08-18:
@mikeller I think we can close this

mikeller, 2016-08-18:
@skaman82 Yes. To conclude, while we not actively discourage use on mobile devices, it is also not something that we have the resources to provide support form

closed 2016-08-18 by mikeller

michelp29, 2016-08-19:
@skaman82,  what is your setting value for :
Settings > System > Display, cursor "Change the size of text, apps and other items" ?

skaman82, 2016-08-19:
@michelp29 it is "125% (recommended)" - never changed that setting