betaflight/betaflight-configurator #218: OSD: Reset Layout Option, Lost Elements after move in Preview

SnappyFPV, 2016-08-13:
Hi i use the Omnibus FC with integrated OSD.

I wish the GUI has a Layout Reset Button. Also some Elements that i have moved by mouse are disapeared in preview - can't see them anymore. Look in screenshot - timer is activated - not visible anymore after movement in preview.

[image]

Tomcatt, 2016-08-16:
So that's what the OSD section looks like.  I need to have a FC with an intergraded OSD on it in order to see this?  I thought my MinimOSD would work.

nathantsoi, 2016-08-16:
yes you need a board with a BFosd like: https://github.com/betaflight/betaflight/wiki/Board---OMNIBUS, mwosd is not supported, yet.

@SnappyFPV, i have fixed the lasted configurator (on master) to make it impossible to position an element off screen. if you need to reset the layout, use the "reset to defaults" on the other status or config page (I think that's where the button is?)

if that works for you, feel free to close this issue

ksFPV, 2016-08-29:
i also got the omnibus FC .. working with rc12 and latest Configurator.

I think the Problem is that if you drag something to the Betaflight Logo Area it's behind the logo and you won't be able to drag it arround anymore since the Logo is in Front.
I have this Problem with my CPU OSD .. it Shows in my Goggles but i can't move it anymore since the Logo is in front of it in the Configurator.

An Reset Switch or a Layer wich keeps your OSD Entrys up front would fix this i guess.

nathantsoi, 2016-08-29:
Good call. I'll add a reset button or a way to show hidden items. We used
to have the position shown as an integer value. Maybe we should bring that
back for now?

ksFPV, 2016-08-29:
maybe just combine the background Picture with the Logo itself and not the Logo as unmoveable object and it should work also i guess ... i'm not that deep into programming.. i just found that the logo is declared as unmovable Object somewhere in the code.

Or maybe an option to hide the logo (i think i've seen that in the Past or in CF Configurator somewhere)

skaman82, 2016-08-29:
I would remove the logo completely. It is not a positionable element.

nathantsoi, 2016-09-23:
It can be hidden now

closed 2016-09-23 by nathantsoi

CheezeFPV, 2020-06-28:
I have found this issue again.  I'm trying to add the "Mah Drawn" but when I turn the switch on nothing appears.

I've tried turning the "everything on" switch, that hasn't helped.

Is there an OSD reset somewhere?  Or will I have to completely reset the whole board? :(