betaflight/betaflight-configurator #223: Button

Tomcatt, 2016-08-16:
Are you able to add a DFU Mode button on the Setup page?  That way you don't have to click on CLI then type it.

Yeah I know, lazy.
LOL

mikeller, 2016-08-16:
This will have to be added to the MSP protocol on the flight controller side first.

borisbstyle, 2016-08-16:
It does that automatically actually once you press flash

gaelj, 2016-08-16:
Yes, what's the point ?
Also could you rename the issue to a more explicit title please ? Such as "add DFU button"

Tomcatt, 2016-08-16:
I was looking for the option to make it a feature or a small enhancement; not a problem.

borisbstyle, 2016-08-16:
Yes but can you explain why it is needee?

Tomcatt, 2016-08-16:
Example
I'd like to go to DFU; so apply power configurator it sees the FC and connects.  First page I see is the setup page.  Oh look, there's "Reboot to DFU".  Why I didn't have to click on CLI to bring up a different page, click on the command box to activate the cursor and hit four keys on the keyboard to do to achieve DFU.

Or for some of us take the top plate off and short the BL pad with one hand and plug the other end of the USB to the computer.  (But that's reserved for if the flash goes bad)

GUI can be as good as CLI but this isn't the time or place for that age old argument.

gaelj, 2016-08-16:
Ok but our question is, why would you want to go in DFU if it's not to
flash (in which case DFU automatically kicks in) ?

Tomcatt, 2016-08-16:
Ah, I see where you two are coming from with that way of doing DFU.  I was going about it a different way.  No worries. My bad.

Could you guys close this ticket please?

gaelj, 2016-08-16:
I can't, but you can close it if you opened it :)

closed 2016-08-16 by skaman82