betaflight/betaflight-configurator #237: 2.7.3 sets airmode_activate_throttle to 70 on any changes in PID Tuning tab

etracer65, 2016-08-21:
Causes air mode to be active immediately on arming and leads to I-term windup and the quad wanting to take off on its own.

borisbstyle, 2016-08-21:
Are you using RC8?

etracer65, 2016-08-21:
No, RC7

etracer65, 2016-08-21:
Just tested with RC8 and the same behavior.  Any changes on the PID Tuning tab cause airmode_activate_throttle to be set to 70.

borisbstyle, 2016-08-21:
Try 1.7.4
https://chrome.google.com/webstore/detail/kdaghagfopacdngbohiknlhcocjccjao/publish-delayed?hl=nl

etracer65, 2016-08-21:
Waiting for the new version to be active - still shows 1.7.3 in the chrome store.  The link you posted gives an error.

Er is een fout opgetreden:

Item niet gevonden. Dit item is mogelijk door de auteur verwijderd.

etracer65, 2016-08-21:
I see the commit though and that looks like it will fix it.

etracer65, 2016-08-21:
1.7.4 just showed up and it fixes the problem.

closed 2016-08-25 by mikeller