betaflight/betaflight-configurator #238: Google Chrome apps are no longer being supported

Austinthemighty, 2016-08-21:
here is a link to the article that state that google is dropping support for google chrome apps: http://wccftech.com/google-planned-support-chrome-apps-mac-windows-linux/ 

In summary;
Google to phase out Chrome apps on desktops means that users will no longer be able to run Chrome apps on any platform except for Chrome OS. Google’s plans start later this year as Chrome apps will only be available to Chrome OS users. Current installed apps will not be affected by this and can still be updated. Moving forward at sometime in the year 2017, the Chrome Web Store will no longer display Chrome apps at any of the above mentioned operating systems. Finally, in 2018, you will not be able to load any of them on Macs, Windows and Linux.

mikeller, 2016-08-21:
Thanks for letting us know. There are already plans under way to do a reimplementation of the configurator based on a more modern framework. The sunsetting of Chrome apps, and the timeline that is planned for this, fit in nicely with those plans.

closed 2016-08-21 by mikeller

Austinthemighty, 2016-08-21:
ok that is good the hear, also can we create an app that works on android and ipad that works via bluetooth

gaelj, 2016-08-21:
GPL is not accepted on the Apple Store.

Austinthemighty, 2016-08-22:
@gaelj you can still do it @borisbstyle just has to claim copyright to it

Austinthemighty, 2016-08-24:
hey can @borisbstyle create a new repository for the code, i was looking into http://nwjs.io/ for clean flight and a lot of things have to change inorder for it to work