betaflight/betaflight-configurator #241: feature request: add ability to control motor output in motors tab via RC throttle stick

csurf, 2016-08-22:
It would seem helpful to be able to control the motor output in motors tab via the RC stick input (over receiver link). 

Perhaps we could add an extra checkbox/binary slider to enable control via RC throttle stick. This would allow for easier & finer control of motor output in certain cases (such as for dynamic balancing); much easier to control the throttle stick on the radio versus having to fiddle with keyboard arrows &|| home/end keys.

Sound reasonable? Doesn't seem like it would be too difficult to implement in  code.

borisbstyle, 2016-08-24:
I dont see a point of doing this while you have sliders already

Austinthemighty, 2016-08-24:
i like the idea, it would be a lot easier when trying to balance the motors, also you would have like a check box to select what motor you want when doing this

mikeller, 2016-08-24:
Honestly, I think we are overthinking this. If you want to use the sticks on your remote to control the motors, just disconnect USB and arm it. ;-)

borisbstyle, 2016-08-24:
I still dont see a usefull case for this. Didnt hear any good argument
about why it should be there.

You can use various ways for driving the sliders.

skaman82, 2016-09-14:
I agree with @borisbstyle and @mikeller

closed 2018-05-13 by stale[bot]