betaflight/betaflight-configurator #254: Just cosmetics: Columns don't match up

RipperDrone, 2016-08-28:
On the BFC PID page, column titles and data fields do not match up at all:
[image]

Is this difficult to implement?

mikeller, 2016-08-28:
It needs some tweaking in the .css, which I am not particularly good at. I was hoping @skaman82 would give the PID tuning tab a makeover now that it has the new filter sub tab.

RipperDrone, 2016-08-28:
Ok, maybe @skaman82 will do the trick sometime. In case: The craft name field could be a little wider to allow for more than 8 characters:

[image]

skaman82, 2016-08-28:
I take a look at this next week if I have time!

mikeller, 2016-08-28:
@RipperDrone: Can you please create a separate issue for the name field size, so we can keep track of things?

RipperDrone, 2016-08-28:
@mikeller Done.

skaman82, 2016-08-29:
done

closed 2016-08-29 by skaman82

mikeller, 2016-08-29:
Dammit, you're too fast, @skaman82 ;-) There will be more changes to the table coming from #252. Sorry.

RipperDrone, 2016-09-01:
Feedback: Name field and basic PIDs look wonderful, however STRENGTH & TRANSITION are still a little mis-scaled:
[image]