betaflight/betaflight-configurator #260: GUI: Menu dead after resizing application window (e.g. horizontally)

RipperDrone, 2016-09-04:
After extending the canvas of the app window horizontally, mouse selection / left-click does no longer select menu items - menu seems dead.

mikeller, 2016-09-04:
What menu? What OS?

RipperDrone, 2016-09-04:
All menu items of BFC itself. OS=Win10 x64. I reverted back to 1.7.7, then resizing windows lets the menu stay alive.

mikeller, 2016-09-04:
There is not really any menus and menu items in the configurator, so I cannot quite follow you.

RipperDrone, 2016-09-04:
sry, did not make myself clear enough. I am referring to the menu icons in the left side bar, e.g. Setup, Ports, Configuration, ...

Those are all dead in the latest build, after resizing the app window.

mikeller, 2016-09-04:
Hmm, I cannot reproduce under Linux. Maybe someone with Windows can have a look?

RipperDrone, 2016-09-11:
Seems fixed in 1.8

closed 2016-09-11 by RipperDrone