betaflight/betaflight-configurator #261: GUI: cloumn alignment still confused?

RipperDrone, 2016-09-04:
@skaman82 : Sry to bother, but with latest commits, still column border alignment seems confused on my Windows 10 system:

[image]

In the VEL/ALT tab, everything matches just perfect - why is it different on the other tables?

mikeller, 2016-09-04:
@RipperDrone: Please be patient. There are more changes to the PID tab in the pipeline, we'll clean up the layout once everything is in.

RipperDrone, 2016-09-04:
no prob, just wanted to inform @skaman82 , since Albert seemed to be working on these alignments with his recent commits (he even referenced to these issues in his commit comments and the issues were put into closed state already - but not fixed yet).

mikeller, 2016-09-11:
This seems to be fixed now, @RipperDrone?

RipperDrone, 2016-09-11:
@mikeller it's better but not fixed. Depending how you size the window horizontally, there is a column mismatch in STRENGTH and TRANSITION still. All others are a perfect match.

closed 2016-09-14 by borisbstyle

RipperDrone, 2016-09-17:
@skaman82 All perfect now - even on Windows 10. Thanks, mate!

skaman82, 2016-09-17:
That wasnt me this time - but still great! ;)

PR #272 (2016-09-14): Fixed more spacing issues on the PID tab.
Fixes #261.