betaflight/betaflight-configurator #284: Chrome applications are deprecated by google

jaimevisser, 2016-09-15:
Betaflight configurator will need to move to either Electron or NW.js if it is to be kept in a usable state when chrome app support ends.

Electron is also used as the basis of Slack and Atom. Quite mature and packaging options for all major platforms.

NW.js is a bit older then Electron and is also widely in use. Packaging is a bit more difficult but the advantage is that it has support for all chrome api's which could make porting a lot easier.

KiteAnton, 2016-09-15:
Yes thanks for the info.  It is already known in the dev team and there will be an independent configurator in the future, stay tuned. :)

closed 2016-09-15 by KiteAnton

Austinthemighty, 2016-09-16:
should we re-open #238 so people see it